RobotecAI/o3de-ros2-gem #78: Spawning robots

adamdbrw, 2022-05-26:
A convenient, generic way to spawn robots in a specified place.

pawellech1, 2022-08-19:
A component responsible for spawning robots needs to be designed and implemented. Spawning should be available both inside the O3DE editor and at runtime.

The component should allow to spawn robot in a place selected by user but it should also be able to specify the right place to spawn robot by itself. It should be possible to spawn multiple robots at the same time. User has to be able to control the robots after they are spawned.

pawellech1, 2022-08-31:
ROS2Spawner Component will store both AzFramework::EntitySpawnTicket and AZ::Data::Asset<AzFramework::Spawnable> as its members. It will be possible to set m_spawnable from the Editor's GUI. User will be able to send commands to ROS2Spawner using ros2 service at runtime. It will be optional for a user to specify where the robot should be spawned.

ROS2RobotSpawnPointsDetector will be a component responsible for finding a suitable spawn point for a robot. It will also allow to check if the place specified by user is suitable for spawning.

After receiving a command with a point of spawn specified by user, ROS2Spawner will check if the point is suitable for a spawn. If so, it will spawn a robot using AZ::Interface<AzFramework::SpawnableEntitiesDefinition>. If the point is not specified in the message, the ROS2Spawner will ask ROS2RobotSpawnPointsDetector for possible spawn points and then spawn a robot in one of them. If the robot is spawned correctly ROS2Spawner responses with a success message. If the robot cannot be spawned for some reason ROS2Spawner responses with a fail message.

Communication between ROS2Spanwer and ROS2RobotSpawnPointsDetector will be via ros2 service. In the first iteration ROS2RobotSpawnPointsDetector will response with fixed coordinates.