RobotecAI/o3de-ros2-gem #144: Design of Vehicle Dynamics feature set

adamdbrw, 2022-08-11:
The goal is to create Components and classes to support simulation of movement of diverse types of robot mobile bases. **The focus in this task should be narrowed down to the type required by the MS2 demo** (a four-wheeled, Ackermann drive type), with only rudimentary analysis of other types to understand what generalizations might be useful for the design. 
This task should include the following items:
* [ ] Domain research, including available open-source solution (PhysX vehicle Gem, others) and solutions in other simulators.
* [ ] Design of components and classes. How would a developer use the feature?

Outcomes:
* [ ] Document capturing useful knowledge of how this feature is designed in existing solutions, comparison of major advantages and weaknesses.
* [ ] First design document
* [ ] Accepted design document, after a review.
* [ ] Finally, a set of issues capturing implementation of the proposed design (some issues can be outside of MS2).

adamdbrw, 2022-09-01:
[VehiclePhysics.pdf](https://github.com/RobotecAI/o3de-ros2-gem/files/9470489/VehiclePhysics.pdf)
First draft for reading.
This will be converted to a PR with docs

adamdbrw, 2022-09-05:
First iteration has been submitted.
Sub-tasks will include:
- Tasks for implementation of proposed components
- Tests on physics scene
- Implementation of a working example for a four-wheeled robot
- Second iteration of design

pawelbudziszewski, 2022-09-06:
It seems fine to me, I have just one question regarding the diagram: why the WheelControllerComponent is connected to the VehicleModel via WheelsConfiguration, not directly? I'm not saying this is wrong, it just seems more natural for me if these wheel-related components are swapped places.

adamdbrw, 2022-09-06:
WheelsConfiguration is a part of VehicleModel which holds (references) to WheelControllerComponents.

pawelbudziszewski, 2022-09-06:
Ok for me

PR #158 (2022-09-13): design docs and a diagram for vehicle dynamics
Fixes #144
closed 2022-09-13 by adamdbrw