RobotecAI/o3de-ros2-gem #163: Improve URDF importer UX

adamdbrw, 2022-09-08:
**Is your feature request related to a problem? Please describe.**
Importer is currently in a very basic state in terms of user experience. We would like to improve user experience when importing robots.

**Describe the solution you'd like**

- [ ] The Importer GUI is neat and does not look like work in progress
- [ ] Typical pathways (things existing, missing, some common errors) are handled well, reducing chance of running into unpleasant behavior.
- [ ] The user is pointed to the Assets directory when importing
- [ ] The robot is spawned in a reasonable place. Consider adding a camera to the robot (overhead) and moving viewport there after the import.
- [ ] After import, inform about heuristics-base choices made by the importer and provide checklist for the user to guide them in the process - such as adding sensors, making sure the robot drives in the simulation, checking materials etc.

**Describe alternatives you've considered**
Importing URDFs automatically. This, however, poses multiple problems since we would like user to be aware of domain mapping issues as well as some heuristic choices we have made during the import (such as deducing drive type and populating vehicle dynamics components based on wheels).

**Additional context**
Create subtask for each item when starting to work on it with this task as a meta-issue. Create issues also for tasks which will be outside of milestone 2 - they will be handled with lower priority.

adamdbrw, 2022-09-22:
After import, a summary and a checklist appears (with links to User Guide):
The URDF for <robot_name> was imported successfully. A prefab has been created in <location>. 
- If user selected "spawn in default point" option: "The robot was spawned in the default point, <point xyz>"
- If RobotControl component was not added successfully (no "base_link" found) -> add item to checklist to add RobotControl, otherwise add "check if RobotControl is configured properly". Both with a link to User Guide.
- If VehicleModel component was not added succesfully (no "base_link" found or no wheels found) - add item to checklist to add VehicleModel and WheelControllers. Otherwise, add "check if VehicleModel and WheelControllers are configured properly". In both cases, add "Adjust PIDs for speed and steering so that your robot can move adequately" and some helpful link as well.
- In any case, add a section "Add required sensors for your Vehicle, you can find available sensors <here -link>"

michalpelka, 2022-12-02:
A lot of work was done in #297 .