RobotecAI/o3de-ros2-gem #165: Support absolute path in URDF assets

michalpelka, 2022-09-08:
**Is your feature request related to a problem? Please describe.**
xacro generates absolute path from xacro file:
Xacro :
```
 <mesh filename="file://$(find my_robot_description)/meshes/base_link.stl" scale="1 1 1"/>
```
URDF:
```
<mesh filename="file:///home/michal/ros2_ws/install/my_robot_description/share/my_robot_description/meshes/base_link.stl" scale="1 1 1"/>
```

**Describe the solution you'd like**
- I recommended to strip path to relative, since the o3de get assets from manager not from command line
- Or, simply provide the user with concise information that it is not supported.

michalpelka, 2022-09-08:
This [PR](https://github.com/RobotecAI/o3de-ros2-gem/pull/156) should include solution for this issue.

michalpelka, 2022-10-12:
Try to replace the absolute piece of the path with `package://` and put meshes to the Asset directory and allow to process by Asset Processor. 
At this moment we are thinking about an elegant way to solve this issue.

danielemorra98, 2022-10-13:
I discovered that the prefix 'package://' stands for 'pwd' so I moved the URDF inside the project folder and the meshes inside Assets/O3DEScene/ (where there are the other .fbx files). With this new setup I don't get any more errors or warnings when loading the URDF but still no sign of meshes.

michalpelka, 2022-12-01:
We merged #297 which has, a robust solution for discovering urdf<-> O3DE mapping of assets. It uses a checksum to find the correct mapping. I've tested the solution against multiple urdf and it works fine. Please check it out.

closed 2022-12-01 by michalpelka