RobotecAI/o3de-ros2-gem #170: Game view for simple shapes in URDF importer

michalpelka, 2022-09-09:
**Is your feature request related to a problem? Please describe.**
The created prefab has simple shapes (like box, cylinder, sphere) created as e.g. `EditorSphereShapeComponent`.
The created has option "Game view" disabled (see attached screenshot), so it is not visible after hitting CRTL+G.
[image]

**Describe the solution you'd like**
I would like to have properly drawn simple shapes in the game mode. Ideally without opacity.

**Additional context**
 **Minimal URDF**:
 ```
 <robot name="test">
    <link name="base">
        <inertial>
        <origin xyz="0. 0. 0."/>
        <mass value="1."/>
        <inertia ixx="1." ixy="0." ixz="0." iyy="1." iyz="0." izz="1."/>
        </inertial>
        <visual>
            <origin rpy="0 0 0" xyz="0 0 0"/>
            <geometry>
                <box size="1 1 1"/>
            </geometry>
        </visual>
        <collision>
            <origin rpy="0 0 0" xyz="0 0 0"/>
            <geometry>
                <box size="1 1 1"/>
            </geometry>
        </collision>
    </link>
</robot>
```

adamdbrw, 2022-09-09:
We won't be using Editor Shape Colliders. This option needs to be set in the Editor Collider instead

michalpelka, 2022-09-09:
Yes, but it is for visual links, not colliders.

adamdbrw, 2023-01-23:
@michalpelka please check if this is still occuring