RobotecAI/o3de-ros2-gem #217: Editor stops working on left-lick on component from this Gem

michalpelka, 2022-09-21:
**Bug description**
Like in the title, the left click on component delivered by this Gem cause Editor to crash.

**To Reproduce**
Steps to reproduce the behavior:
1. Open sample project (e.g.  Ros2WarehouseDemo)
2. Open level
3. Select robot base link
4. Left click on component (e.g. ROS2 Frame)
5. Editor stops

**Expected behavior**
Show Editor's context menu

**Screenshots / videos**
If applicable, add screenshots/videos to help explain your problem.

**Test environment**
 - OS: [e.g. Ubuntu 20.04]
 - ROS: [e.g. Galactic]

**Additional context**
The danger zone is marked with red frame :
[image]

adamdbrw, 2022-09-22:
More details found by @michalpelka:
The crash is in 
`Code/Framework/AzToolsFramework/AzToolsFramework/UI/PropertyEditor/EntityPropertyEditor.cpp`
```
 AzToolsFramework::Components::EditorComponentBase* editorComponent = static_cast<AzToolsFramework::Components::EditorComponentBase*>(component);

AZ_Printf("FOOBAR", "editorComponent %ld \n", editorComponent->GetId());
if (editorComponent)
{
   editorComponent->AddContextMenuActions(&menu);
}
```

michalpelka, 2022-09-22:
The whole situation looks like we have a problem in polymorphism that is visible in runtime.
The code crashes in ` EntityPropertyEditor::RequestPropertyContextMenu`. 
The logic there is like that:
- Our component (e.g. `ROS2FrameComponent`) is taken here. 
- It is static-casted to various parent types.
- Casting to `AZ::Component` works fine
- Casting to `AzToolsFramework::Components::EditorComponent` is performed gives a pointer
- This pointer however is not valid since ROS2FrameComponent do not inherit from `AzToolsFramework::Components::EditorComponent`
- The whole program crashes down the road `editor component->AddContextMenuActions(&menu);`  since the program tries to do something on an invalid piece of memory given udefined behavior (crash luckily).
One more note:
- the whole thing works fine when we switch to EditorComponent like [here ](https://github.com/RobotecAI/o3de-ros2-gem/pull/218/commits/7cf28af74b6f7a57f4e1ef43e5410fdda33084b2#)

michalpelka, 2022-09-23:
I can see that URDF importer creates Components in the wrong way. When components are created in Editor, they are handled correctly by `Code/Framework/AzToolsFramework/AzToolsFramework/UI/PropertyEditor/EntityPropertyEditor.cpp` 
I will fix that in URDF importer.

adamdbrw, 2022-09-23:
Perhaps it is best to add a utility function which encapsulates all that

michalpelka, 2022-09-23:
Yes, the change will be a little large - it seems that we cannot pass parameters that we used to pass with a constructor.
A good example here is 
````
        ControlConfiguration controlConfiguration;
        controlConfiguration.m_robotConfiguration.m_body = rootEntityId;
        controlConfiguration.m_broadcastBusMode = false;
        controlConfiguration.m_topic = "cmd_vel";
        AZ::Entity* rootEntity = AzToolsFramework::GetEntityById(rootEntityId);
        rootEntity->CreateComponent<ROS2RobotControlComponent>(controlConfiguration)
 ```
 Should we expose those all parameters to use with EBuses?
 The second question considers components that are already EditorComponents like:
```
entity->CreateComponent<PhysX::EditorRigidBodyComponent>(rigidBodyConfiguration, physxSpecificConfiguration);
```
Those often expose crucial features with their constructors.

adamdbrw, 2022-09-23:
Wouldn't it be
CreateComponent (as now)
Do some additional things
?

closed 2022-09-28 by adamdbrw