RobotecAI/o3de-ros2-gem #236: `cmd_vel` topic from RobotControl subscribes before the simulation is played

pijaro, 2022-09-28:
**Bug description**
`cmd_vel` topic from RobotControl subscribes before the simulation is played if the robot prefab is present in the scene.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a robot prefab with a Control
2. Insert prefab in the scene
3. Save the scene
4. Restart O3DE
5. Check `ros2 topic list`

**Expected behavior**
The topic is not subscribed before the scene launches.

**Test environment**
 - OS: e.g. Ubuntu 20.04
 - ROS: Galactic

**Additional context**
I can't reproduce it with just a single script, it is happening when I use a full robot prefab.

pawellech1, 2022-11-03:
I cannot reproduce it, the cmd_vel topic does not appear in `ros2 topic list` output before or after the simulation. The only topics that are in this output are: 
- /clock
- /parameter_events
- /rosout
- /tf
- /tf_static

I think the issue might be solved.

adamdbrw, 2022-11-03:
The Component has been reworked and the subscription handling changed. It is quite possible the issue is no longer there.

closed 2022-11-03 by adamdbrw