RobotecAI/o3de-ros2-gem #238: Asset Processor Asserts when starting up Batch

alek-kam-robotec-ai, 2022-09-28:
**Bug description**
When starting up Asset Processor from Batch, the Asset Builder's Class Builder asserts with a "You can attach attributes only to UiElements!" message.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
The Asset Processor should run without any asserts triggered.

**Screenshots / videos**
[image]

**Test environment**
Not specified.

adamdbrw, 2022-09-28:
This comes from ROS2SensorComponent reflection apparently.
But the only Attribute there is:
`->Attribute(AZ::Edit::Attributes::Category, "ROS2")`
Not sure what is wrong with this.

adamdbrw, 2022-09-28:
@spham-amzn

adamdbrw, 2022-09-28:
What is special about `ROS2SensorComponent`? My guess it is that it is the only base component (for `ROS2LidarSensorComponent`, for example). Base class has also this attribute specified. Maybe this is not legal?

adamdbrw, 2022-11-03:
This one is fixed through removing of extra classification

closed 2022-11-03 by adamdbrw