RobotecAI/o3de-ros2-gem #275: Robot Importer failed to load rightly URDF

danielemorra98, 2022-10-12:
**Bug description**
When I am importing a robot model from URDF file, the O3DE engine crashes every time in the <collision> box there is a geometry which measures are less than 1. It seems related to float numbers not correctly loaded (it rounds down floats, e.g. 0.5 becomes 0.0). This problem also involves also the geometry of <visual> tag.

**To Reproduce**
Importing [this URDF](https://github.com/danielemorra98/limo_ros/blob/master/limo_description/urdf/sample.urdf) brings the engine about a crash.

**Screenshots / videos**
The resulting call stack:
[image]

**Test environment**
 - OS: Ubuntu 20.04
 - ROS2: Galactic

adamdbrw, 2022-10-12:
@danielemorra98 it looks like a locale issue. Did you get a warning from the Importer? Could you share the logs?

Please note that urdfdom does not respect locale when parsing URDF. (E.g. if you have 0,5 in URDF, it is loaded as 0). 
If you install ros2 according to instruction and set locale, this should not be an issue.

piotr-zyskowski-rai, 2022-10-12:
@danielemorra98 

Additionaly please share output of `locale` shell command. You should have output similar to this:
```
$ locale
LANG=en_US.UTF-8
LANGUAGE=
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_PAPER="en_US.UTF-8"
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT="en_US.UTF-8"
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=en_US.UTF-8
```
For the URDF parsing case, `LC_NUMERIC="en_US.UTF-8"` is crucial

adamdbrw, 2022-10-12:
FYI URDF importer is a kind of "Early Access" feature and has a lot of edges to be polished still. If you would like to help improving it, that would be great! Some issues are already listed.

lgleim, 2022-10-12:
FYI - the PhysX crash when providing 0 as input to the `SetCylinderRadius` method was fixed two days ago in this commit: https://github.com/o3de/o3de/commit/147cd7460283fdb9c6d154d266506309a0efa808
Given the right locale, both issues should hopefully be handled. 

@piotr-zyskowski-rai Would it make sense to create a console warning whenever URDF parsing is conducted without `en_US.UTF-8` locale?

adamdbrw, 2022-10-12:
@lgleim glad the change is in o3de. The warning should be there already with https://github.com/RobotecAI/o3de-ros2-gem/pull/197

danielemorra98, 2022-10-12:
Thanks guys. In my locale settings there was:
`LC_NUMERIC=it_IT.UTF-8`
causing the issue.

closed 2022-10-12 by danielemorra98