RobotecAI/o3de-ros2-gem #282: No loaded mesh when importing URDF

danielemorra98, 2022-10-13:
**Bug description**
When I load the URDF model, the meshes related to the <visual> tags are not loaded. The path of my files are the same as it is described [here](https://github.com/RobotecAI/o3de-ros2-gem/issues/165#issuecomment-1277287346), and I do not get any errors or warnings from the _Asset Processor logs_ . 

**Test environment**
 - OS: Ubuntu 20.04

adamdbrw, 2022-10-13:
since paths are changed to relative by the importer, they need to be correct after such change. Make sure your urdf and meshes directory follow this structure.

danielemorra98, 2022-10-14:
I think that the AssetProcessor cannot load .dae mesh files; when I tried with .fbx files it went rightly.

adamdbrw, 2022-10-14:
You need to enable it through a setting

danielemorra98, 2022-10-17:
I tried with these settings and I obtained the followinf WARNING msgs:
[image]
Unfortunately still no meshes loaded and I don't think it is a problem of the mesh since Blender loads it correctly.

adamdbrw, 2022-11-03:
@danielemorra98 the meshes are likely loaded but 1/100th of size. I mentioned the scale patch.

michalpelka, 2023-01-20:
Closing - this issue was resolved with the introduction of #297.

closed 2023-01-20 by michalpelka