RobotecAI/o3de-ros2-gem #293: Imported robot is not static

michalpelka, 2022-10-24:
**Bug description**
I've tried to import a robot description from this file : 
https://github.com/husky/husky/blob/foxy-devel/husky_description/urdf/husky.urdf.xacro

**To Reproduce**
Steps to reproduce the behavior:
1. Checkout repository from Clearpath
2. Build ros2 workspace
3. Generate `husky.urdf` from husky.urdf.xacro with 'xacro'
4. Adjust path in `husky.urdf` to relative and copy to assets catalog 'mesh'
5. Import URDF
6. Start the simulation

**Expected behavior**
The robot should be dynamic, it seems to be static.

**What happened**
The structure of URDF is as follows:
[image]
The `base_link` node does not have a rigid body component. Its child called 'inetrial_link' has it instead. After adding rigid body component with a default config the whole robot becomes dynamic

**Test environment**
 - OS: e.g. Ubuntu 22.04
 - ROS: Humble

**Additional context**
Add any other context about the problem here.

danielemorra98, 2022-10-24:
I had the same issue and I found out that if you don't specify **inertial** characteristics to your _base_link_ o3de doesn't attach a rigid body component at that link and it won't be subject to gravity

michalpelka, 2022-10-24:
Thanks for your valuable input @danielemorra98 . My explanation is like this:
 - joints are not created in game mode since they do not have valid lead joint
 - the real inertial body has no displayed mesh attached, it is simulated by physics, but there is no way to update the displayed scene with the result.
I think that URDF importer needs to take into account the limitations and translate the imported URDF graph into one that is fit for o3de.

adamdbrw, 2023-02-06:
This should be raised up to the user, since URDF lacks the information. For example:

" <Error message>. To resolve this issue, consider the following steps:
Add inertial information to base_link in URDF and import again.
Alternatively, add RigidBodyComponent to the imported model in the base_link entity. Adjust parameters to reflect the real robot inertia.
"