RobotecAI/o3de-ros2-gem #309: Standards and styles / File Structure & Contents

adamdbrw, 2022-11-09:
For the Copyright header:
- [ ] [Copyright Header Requirements](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#copyright-header-requirements)- **do not** follow this guideline yet. Wait for resolution of ongoing discussion about attribution in headers. 

For the API definitions, follow:
- [ ] [API definitions](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#api-definitions)

michalpelka, 2022-12-02:
I have a problem with the second task here. 
In standard there are clear statements that we should try to limit the usage of types:
```
Minimize type dependencies on APIs where possible
```
We can address that by using forward declaration and switch from reference to pointers.
That can be done for `rclcpp::QoS`:
- https://github.com/RobotecAI/o3de-ros2-gem/blob/development/Code/Include/ROS2/Communication/QoS.h
- https://github.com/RobotecAI/o3de-ros2-gem/blob/development/Code/Include/ROS2/Communication/TopicConfiguration.h

For `geometry_msgs/msg/transform_stamped.hpp`:
- https://github.com/RobotecAI/o3de-ros2-gem/blob/development/Code/Include/ROS2/Frame/ROS2Transform.h

# Problematic files
The forward declaration is not a silver bullet, it is a bad idea for following files.
## https://github.com/RobotecAI/o3de-ros2-gem/blob/development/Code/Include/ROS2/RobotControl/ControlSubscriptionHandler.h
It is template, it has implementation. It has to be dependent to rclcpp headers.

## https://github.com/RobotecAI/o3de-ros2-gem/blob/development/Code/Include/ROS2/Utilities/ROS2Conversions.h
It is part of API that covers rclcpp types to o3de and _vice versa_.  It has to be dependent on rclcpp headers.

## https://github.com/RobotecAI/o3de-ros2-gem/blob/development/Code/Include/ROS2/ROS2Bus.h
It allows getting a handle to the ROS2 node.  It has to be dependent on rclcpp headers.

In problematic files, API is not following requirements and there is no feasible way to fix it. I think although it is not a problem since this API is not intended to O3DE developer. 
According to https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#api-definitions
```
The primary consumer of the API reference are developers using O3DE and reading the comments directly in source code.
```
I disagree, with that statements, this API will be consumed by ROS developer who wants to extend their simulation beyond what Gem's code has. It has next to nothing to offer for O3DE developers. With that, I think we should omit requirements to minimize types of dependency.
@adamdbrw @spham-amzn please comment here, thanks!

spham-amzn, 2022-12-09:
> I have a problem with the second task here. In standard there are clear statements that we should try to limit the usage of types:
>
> ```
> Minimize type dependencies on APIs where possible
> ```
>
> We can [...]

```O3DE Developer``` would also include ROS developers if they wish to extend their simulation. The guidance was meant to reduce build dependencies when possible. For the problematic files mentioned here, I think its find to omit this requirement then rather than try to something un-natural to try to conform to this guidance. Thats where the `where possible` part of the guidance comes in

closed 2023-01-23 by adamdbrw