RobotecAI/o3de-ros2-gem #313: Standards and Styles / API Reference

adamdbrw, 2022-11-09:
Ensure the Gem API follows the following guidelines:
https://github.com/o3de/sig-core/blob/main/governance/API-Ref-Guidelines-Update.md

Note that these should be in place already for the code comments so it should be a bit less work - hence not dividing this task further.
If something turns out to be more effort, additional task could be produced.

adamdbrw, 2022-12-01:
@spham-amzn I did quite some improvement for this task with #332, possibly there is not much more to do but I will leave it for you to decide.

spham-amzn, 2022-12-02:
@adamdbrw I've done a pass on the https://github.com/o3de/o3de-extras/pull/70 related to this task. Please look over when you get a chance, and once all the comments are resolved, I can close this issue

closed 2022-12-07 by spham-amzn