RobotecAI/o3de-ros2-gem #314: Migration / Create a PR to O3DE repository

adamdbrw, 2022-11-09:
- [ ] Initialize empty repository
- [ ] Submit a PR of the entire development branch, preserving the history

forhalle, 2022-11-29:
@adamdbrw - My understanding is all of the review work on this board is blocked by this task. Can you please confirm when you will be able to complete it? Please let me know if this needs to be assigned to someone other than you. Thanks!

adamdbrw, 2022-12-01:
For now, we have been applying guidelines and fixes to the codebase (as opposed to the migration PR). The steps that I am following right now are:
1. Stabilization
   1. We had issues with O3DE development branch, this is resolved (for now).
   1. Fixing known issues within the Gem / Demo (not related directly to migration).
1. Applying our assigned cleanup / style tasks (#310, #311) already.
   1. These are now merged into development and moved to review, and can be checked when the PR is cast.
1. Improving URDF import from raw prototype to manageable prototype (so we can say this is actually a feature).
1. Doing the #313 work - this is merged and has "In review" status as it needs to be checked in the PR.

Next steps:
1. Set main branch to development (but also resolve the LFS issue #322) in the process.
1. Determine the cutoff moment (existing feature PRs will be held until after the migration and migrated to the new repo before being reviewed).
1. Cast the Gem PR targeting o3de/o3de-extras repository, using the new main branch.
1. Inform that it is ready to review / migration process.

closed 2023-01-23 by adamdbrw