RobotecAI/o3de-ros2-gem #316: Migration / Copyright headers

adamdbrw, 2022-11-09:
Due to migration into a new space, it would be difficult to see who developed the Gem (aside from analyzing commit history).
Apache 2.0 license requires proper attribution, but also it would be nice for Robotec.ai if their contribution was clear.

Some ideas for headers:

1.

```
/*
 * Developed by Robotec.ai for Amazon, 2021-2022. Copyright Amazon.
 * For complete copyright and license terms please see the LICENSE at the root of this distribution.
 *
 * SPDX-License-Identifier: Apache-2.0 OR MIT
 */
```

2.

```
/*
 * Developed by Robotec.ai for Amazon, 2021-2022. 
 * Copyright Amazon.
 * For complete copyright and license terms please see the LICENSE at the root of this distribution.
 *
 * SPDX-License-Identifier: Apache-2.0 OR MIT
 */
```

3.
```
/*
 * Copyright Robotec.ai, 2021-2022.
 * Copyright Amazon, 2021-2022.
 * For complete copyright and license terms please see the LICENSE at the root of this distribution.
 *
 * SPDX-License-Identifier: Apache-2.0 OR MIT
 */
```

In my opinion, number 3 would be the most standard and similar to ros2 attributions (AFAIK copyright and developed by are the same for Apache 2.0, but I am not an expert on this).

forhalle, 2022-11-10:
I followed up with Legal today, and will respond as soon as I hear something. Thank for your patience!

forhalle, 2022-11-16:
Hi @adamdbrw. 

Amazon Legal completed their research on your request to provide attribution on the ROS 2 and demo work and concluded that the O3DE community does not provide attribution or direct credit to contributors or developers of contributions. Amazon doesn’t want to go against the norm, as it could be viewed as Amazon exercising undue influence. We cannot guarantee anything other than the standard header will be accepted for merging, and therefore suggest leaving the original headers as-is with no additional copyright or attribution information for either Amazon or Robotec.ai  – i.e.:

/*
* Copyright (c) Contributors to the Open 3D Engine Project.
* For complete copyright and license terms please see the LICENSE at the root of this distribution.
*
* SPDX-License-Identifier: Apache-2.0 OR MIT
*
*/

I realize this is not the response you were hoping to hear, but I hope you take comfort in the fact that Amazon is also not pursuing copyright or attribution themselves. I am happy to discuss any concerns you have with this at our meeting tomorrow.

Thanks,
Halle

j-rivero, 2022-11-21:
Some ideas that might help to report the attributions in open source projects:

 * The good old AUTHORS (or similar) file: [google opensource description]( https://opensource.google/documentation/reference/releasing/authors) and an example from [OpenSSL](https://github.com/openssl/openssl/blob/master/AUTHORS.md.).
 * Sometime, the README just honor the founder or relevant people in previous phases of development. See [ruby project example]( https://github.com/ruby/ruby/blob/master/README.md?plain=1#L88-L92).
 * In a more formal way, following the [SPDX format](https://spdx.dev/). The format is complex and have many different files, I assume following this thread that setting the real copyright statements is not a way to go so maybe using one of the fields [reserved for comments ](https://spdx.github.io/spdx-spec/search.html?q=comment)can serve to introduce the attribution to RobotecAI.

Hope it helps.

adamdbrw, 2022-12-01:
@forhalle what do you think about @j-rivero ideas? Is there something you find the most suitable among his suggestion?

forhalle, 2022-12-02:
@adamdbrw  - So sorry for the delay. @amzn-pratikpa had some ideas on this, but I didn't get a chance to catch up with him prior to he leaving for PTO. He will be back on Tuesday, and I will work with him to respond at that time. Thank you for your patience. I am confident we will work something out that everyone will be happy with.

amzn-pratikpa, 2022-12-07:
@adamdbrw - My idea was the same as @j-rivero. An "authors.md" which contains the information or just a section at the end of the README. Would either of these work for you?

adamdbrw, 2022-12-07:
@amzn-pratikpa certainly works for me! Let me submit the PR to address this.

adamdbrw, 2022-12-07:
I opted for a short mention at the end of the README.
@forhalle  @amzn-pratikpa please take a look if this looks correct for you:
https://github.com/RobotecAI/o3de-ros2-gem/pull/353

Possible alternatives (please let me know if you prefer as it is now in the PR or with the changes as listed below:
"Developed by Robotec.ai and AWS Game Tech" (instead of "in cooperation with")
"AWS Game Tech" --> "AWS"

closed 2023-01-23 by adamdbrw