RobotecAI/o3de-ros2-gem #322: Broken Git LFS references

lgleim, 2022-11-21:
**Bug description**
The repository has broken Git LFS references. 

**To Reproduce**
1. Check out the repository
2. Run `$ git lfs fetch --all` and see error message 
  ```
  fetch: 3 objects found, done.
  fetch: Fetching all references...
  [8d438c48dc194710dbec4b70245b0d34a2887124fb0e2b2bd1c49c03fd99a575] Object does not exist on the server: [404] Object does not exist on the server
  [7ac9dd09bde78f389e3725ac49d61eff109857e004840bc0bc3881739df9618d] Object does not exist on the server: [404] Object does not exist on the server
  error: failed to fetch some objects from 'https://github.com/RobotecAI/o3de-ros2-gem.git/info/lfs'
  ```

One of these is `(missing) preview.png (7ac9dd09bde78f389e3725ac49d61eff109857e004840bc0bc3881739df9618d)`

**Expected behavior**
No errors

adamdbrw, 2022-12-01:
@lgleim we are migrating to o3de/o3de-extras. The issue will be resolved during the process.

michalpelka, 2023-01-20:
We moved to https://github.com/o3de/o3de-extras and resolved this problematic file.

closed 2023-01-20 by michalpelka