RobotecAI/o3de-ros2-gem #324: URDF importer does not preserve the hierarchy of entities

pawelbudziszewski, 2022-11-21:
**Bug description**
Previously the URDF importer used to preserve the hierarchy of objects connected by joints and recreate this hierarchy in the O3DE entity tree. Now it inserts such objects on one level.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone and build the [ROSConDemo](https://github.com/o3de/ROSConDemo/tree/urdf_update), branch `urdf_update`.
2. Follow the first 3 steps of [this tutorial](https://github.com/o3de/ROSConDemo/blob/urdf_update/docs/URDF_import/HOWTO_import_urdf.md)
3. Open the `apple_kraken_new` prefab

You'll see something like this (with all entities containing rigid bodies on one level):

[image]

**Expected behavior**
It should look like this:

[image]

**Test environment**
Tested on:
- ROSConDemo [fdc70b197e6a43e099305dee08852a00714f522c](https://github.com/o3de/ROSConDemo/commit/fdc70b197e6a43e099305dee08852a00714f522c)
- O3DE [138e4a70a81770bd59fa7972c138e324c55441bd](https://github.com/o3de/o3de/commit/138e4a70a81770bd59fa7972c138e324c55441bd)
- o3de-ros2-gem [d5a76d2857e8f995191be8a4ac006e30db003f69](https://github.com/RobotecAI/o3de-ros2-gem/commit/d5a76d2857e8f995191be8a4ac006e30db003f69)

michalpelka, 2022-11-21:
The hierarchy is created as flat, next it is established with `AZ::TransformBus::Events::SetParent`. Seems that `AZ::TransformBus::Events::SetParent` is ignored.  [Context](https://github.com/RobotecAI/o3de-ros2-gem/blob/development/Code/Source/RobotImporter/URDF/URDFPrefabMaker.cpp#L128)

michalpelka, 2022-11-22:
fixed with #325

closed 2022-11-22 by michalpelka