RobotecAI/o3de-ros2-gem #360: Nav Stack: Review code for O3DE Standards

forhalle, 2022-12-09:
Standards and styles / Naming / Part I:
- [ ] [Class Names](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#class-names)
- [ ] [File names](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#file-names)
- [ ] [Type names](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#type-names)
- [ ] [Variable names](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#variable-names)
- [ ] [Constant names](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#constant-names)
- [ ] [Function names](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#function-names)
Note: Be mindful of: 
https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#exceptions-to-naming-rules
Standards and styles / Naming / General:
- [ ] Make sure that names for classes, variables and functions are sensible. Follow the guidelines: https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#in-general

Standards and styles / Naming / Part II:
- [ ] [Namespace names](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#namespace-names)
- [ ] [Enums](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#enums)
- [ ] [Macro names](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#macro-names)
- [ ] [Abbreviations](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#abbreviations)
- [ ] [Out parameter names](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#out-parameter-names)
Note: Be mindful of exceptions:
[Exceptions to naming rules](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#exceptions-to-naming-rules)

Standards and styles / Header Files
- [ ] [Include guards](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#include-guards)
- [ ] [Include paths](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#include-paths)
- [ ] [Use forward declarations to minimize header file dependencies](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#use-forward-declarations-to-minimize-header-file-dependencies)
- [ ] [Function inlining](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#function-inlining)
- [ ] [Minimize code in headers](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#minimize-code-in-headers)
- [ ] [Including headers](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#including-headers)

Standards and styles / File Structure & Contents:
- [ ] For the Copyright header: [Copyright Header Requirements](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#copyright-header-requirements)
- [ ] For the API definitions, follow: [API definitions](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#api-definitions)

Standards and styles / Classes:
- [ ] [Default constructors](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#default-constructors)
- [ ] [Member initialization](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#member-initialization)
- [ ] [Declaration order](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#declaration-order)
- [ ] [Constness](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#constness)
- [ ] [Override](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#override)
- [ ] [Final](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#final)

Standards and styles / Documentation:
- [ ] [Self-documenting code is better than documenting the code yourself](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#self-documenting-code-is-better-than-documenting-the-code-yourself)
- [ ] [Use full sentences and avoid abbreviations](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#use-full-sentences-and-avoid-abbreviations)
- [ ] [Leave your feelings out of it](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#leave-your-feelings-out-of-it)
- [ ] [Use non-gendered pronouns](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#use-non-gendered-pronouns)
- [ ] [Explaining "why" is often more valuable than "what"](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#explaining-why-is-often-more-valuable-than-what)
- [ ] [Private members need love too](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#private-members-need-love-too)
- [ ] [Redundant documentation is redundant](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#redundant-documentation-is-redundant)

Standards and Styles / API Reference:
- [ ] Ensure the Gem API follows the following guidelines:
https://github.com/o3de/sig-core/blob/main/governance/API-Ref-Guidelines-Update.md
Note: These should be in place already for the code comments so it should be a bit less work - hence not dividing this task further.
If something turns out to be more effort, additional task could be produced.
Standards and styles / Miscellaneous / Part I:
- [ ] [Use AZ or AZStd](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#use-az-or-azstd)
- [ ] [Use of auto](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#use-of-auto)
- [ ] [Use of nullptr](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#use-of-nullptr)
- [ ] [C++ Exceptions](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#c-exceptions)

Standards and styles / Miscellaneous / Part II
- [ ] [RTTI](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#rtti)
- [ ] [Casting](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#casting)
- [ ] [Use of const](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#use-of-const)
- [ ] [Don’t leave disabled code in the source](https://github.com/o3de/sig-core/blob/main/governance/Coding-Standards-and-Style-Guide.md#dont-leave-disabled-code-in-the-source)

michalpelka, 2022-12-12:
Here is the PR with the navigation state (RO2 packages) migrated : 
https://github.com/o3de/ROSConDemo/pull/202 
This PR is created from main branch of this repository: https://github.com/RobotecAI/o3de_kraken_nav

michalpelka, 2022-12-12:
PR with progress on documenting navigation code :
https://github.com/o3de/ROSConDemo/pull/211

spham-amzn, 2022-12-12:
> @forhalle The navigation stack is **not O3DE code, it is ROS 2 code**.
> 
> It does not use anything from O3DE, link to O3DE and so on. I do not believe many of these are applicable. ROS2 has its own standards, please see https://docs.ros.org/en/rolling/The-ROS2-Project/Contributing/Code-Style-Language-Versions.html#id2.
> 
> I believe that having ROS code in O3DE standard **could be confusing for ROS developers** (who are used to ROS standards). There are some items in this task which can be applied (are agnostic to ROS standards), but some of them are in opposition (for example, naming of variables, snake_case vs camelCase). Moreover, things like the use of AZStd libraries do not apply as these libraries are not available for this ROS code at all.
> 
> I suggest to have documentation and license headers as per O3DE standards, but otherwise apply ROS 2 standards to ROS code in the package. What do you think?

The Nav Stack is Python code, and the oinly requirements for python code is PEP-8 and the copyright attribution, I do not think the list above applies

michalpelka, 2023-01-20:
Revied code to ROS2 standards and merged.

closed 2023-01-20 by michalpelka