RobotecAI/o3de-ros2-gem #366: Poor PhysX performances

danielemorra98, 2022-12-16:
**Bug description**
I have a scene with a 4-wheels vehicle on a simple box serving as a ground. The vehicle is controlled through Ackermann Controller. I have have two different setup (from now on COMPUTER-A and COMPUTER-B) in which I run the same simulation with the following difference:

COMPUTER-A has Ros2-Gem checked out at de2d1cdfa68310c6c0d885d65c0d62fa972a4a69 commit (1 dec)
COMPUTER-B  has Ros2-Gem checked out at ba6a13a0f013f7c58bf5039ea08cca391014f09a commit (10 oct)

With COMPUTER-A when I move the robot (through a custom teleop_twist_keyboard node) it does not look smooth at all but it is  bouncing on the ground. That is something that does not happen with COMPUTER-B 

**Test environment**
Both COMPUTER-A and COMPUTER-B:
 - OS: Ubuntu 20.04
 - ROS: Galactic

COMPUTER-A:
 - CPU: Intel Xeon Gold 5218 (64 Cores)
 - GPU: NVIDIA RTX A5000

COMPUTER-B:
 - CPU: 11th Gen Intel Core i9-11900K
 - GPU: NVIDIA RTX A5000

michalpelka, 2022-12-16:
Thanks for raising the issue, could you observe the difference in frames per second between computers A and B?
We did some minor changes to VehicleDynamics, but it is not expected to affect the way torque is applied. 
Could you provide me FPS and videos on robot behavior on Computer-A?

danielemorra98, 2022-12-19:
FPS:
- Computer-A: 7-10 fps
- Computer-A: 30-37 fps 
I was wrong with the GPU card mounted on Computer A. I edited the first message with the right one.

michalpelka, 2022-12-19:
Thank you for your reply.
Current vehicle dynamics need more than 20 fps to work properly.
Correct me if I am wrong, but you are getting such bad performance on Computer-A after activating the vehicle dynamic?

adamdbrw, 2023-01-23:
@danielemorra98 if you run any of the Demos (Vacuum, Agricultural one) and disable VehicleDynamics components, does this affect your fps?
Could you please make a comparison and perhaps add a profiler output to help us identify the issue? We have not experienced performance issues with Vehicle Dynamics previously.