RobotecAI/o3de-ros2-gem #370: Robot Importer cannot load urdf file with .fbx meshes

hhaofan, 2023-01-13:
**Bug description**
We need to use urdf with fbx format meshes to be able to apply materials within O3DE. But the 'robot importer' can not successfully import [urdf](https://github.com/hhaofan/urdf_fbx_minimum/blob/main/iiwa1_fbx.urdf) when it includes a [.fbx](https://github.com/hhaofan/urdf_fbx_minimum) mesh type instead of a [.stl](https://github.com/hhaofan/urdf_fbx_minimum), saying no CRC found.

**To Reproduce**
Steps to reproduce the behavior:

- Go to 'Robot Importer' and import the iiwa1_fbx.urdf file which uses EE_arm.fbx meshes (Files are included in the attachement)

- See error. And if we create the prefab regardlessly, we can see no meshes for the create entity

[image]

**Expected behavior**
Mesh properly imported in project without error reported, given that the format is supposed to be supported according to the code shared at:
https://github.com/RobotecAI/o3de-ros2-gem/blob/development/Code/Source/RobotImporter/Utils/SourceAssetsStorage.cpp#L35
**Screenshots / videos**

**Test environment**
 - OS: Ubuntu 20.04
 - ROS: Galactic

**Additional context**
- When I try to use Blender to view the EE_arm.fbx mesh, it works well, proving the fbx file has no problem.

[image]
 
- If I add a single entity in the editor and include a mesh component using the same .fbx mesh file previously used for the URDF, then it looks how it is supposed to in the scene. This proves that **O3DE** properly supports the conditions for the .fbx mesh file we are using.

[image]

- When I try to import the urdf file with robot importer including the same fbx mesh file, it doesn't import successfully, giving the following error:

[image]

- With the same urdf file, same mesh but converted to the [EE_arm.stl](https://github.com/hhaofan/urdf_fbx_minimum) format instead (files included in the attachement), the robot importer loads the mesh successfully, proving the urdf file and the mesh data are not the source of the problem.

[image]

michalpelka, 2023-01-13:
Thanks for raising that issue.
I've tried to reproduce your error, and partly succeed.  I've cloned your [repo](https://github.com/hhaofan/urdf_fbx_minimum) to my Asset folder. 
Then I imported the URDF and fbx was attached correctly. However, asset processor did not process 'EE_arm.stl' - it is caused the product asset with the same name was already generated from "EE_arm.fbx"

Please verify:
 - if your FBX file was correctly processed by asset processor.
 - please send me 'Project/user/log/Editor.log" after failed import.

hhaofan, 2023-01-20:
Hi Michal,

Thanks so much for your answer. I have tried several times more and also asked my colleages to give a trial, but we all cannot import the urdf file with the .fbx meshes, even if we deleted the .prefab file which is created by the urdf with .stl mesh.

Here is the reorganized [git repo](https://github.com/hhaofan/urdf_fbx_minimum/tree/updated) (switch to the update branch).That might be easier to reproduce.
And here is the answer to your question:

- the asset processor shows my FBX file is correctly processed.

- [Here1](https://github.com/hhaofan/urdf_fbx_minimum/blob/main/Editor_VersionF.log), [here2](https://github.com/hhaofan/urdf_fbx_minimum/blob/main/Editor_VersionC.log), [here3](https://github.com/hhaofan/urdf_fbx_minimum/blob/main/Editor_VersionD.log) are part of the "Project/user/log/Editor.log" file after failure import of urdf with .fbx meshes

michalpelka, 2023-01-25:
@hhaofan in your logs I can see that no fbx file was not picked, are you sure that you are the newest version from o3de-extras? 
Secondly in your [git repo](https://github.com/hhaofan/urdf_fbx_minimum/tree/updated) I can see that you have the assets with the same base-name (EE_arm.fbx and EE_arm.stl). 
Such a situation is problematic in o3de since it will generate product assets from the first file, and fail from the second. 
Please :
 - give other base names for source assets e.g. EE_arm.fbx and EE_arm_collision.stl
 - verify if all source assets were built by Asset Processor.
 - verify if you are using the correct version from o3de-extras.
 Thanks, Michał.

hhaofan, 2023-02-06:
@michalpelka  Hi Michal, 
I tried to 
- rename the .fbx and .stl meshes to different name, but it still can only process the .stl meshes
- all source assets were successfully built by asset processor
- I tried to build it with the latest o3de-extras, but the projects cannot be built successfully with the latest o3de-extra version. It gives some error as [error_log1(Ros2WarehouseDemo Project) ](https://github.com/hhaofan/urdf_fbx_minimum/blob/main/CMakeProjectBuildError_Warehouse.log)and [error_log2(o3de-physics-test-scene Project)](https://github.com/hhaofan/urdf_fbx_minimum/blob/main/CMakeProjectBuildError_physics_test_scene.log). So I am wondering which o3de and o3de-extra commit are you using when you can both successfully build project, and can successfully load the .fbx meshes.  
- P.S. My current o3de commit is `9cd688b15c93ebbec9d18e273035ddf939fa6126`, o3de-extra commit is `9b63103ab9f9ba5cb0f31645070828a97fdc8df7`.

Thanks for your kind help!